s-nakaoka/choreonoid #165: 並進間接の可動範囲の設定について

YudaiTogawa, 2017-09-19:
並進間接の可動範囲を設定するために、jointRangeで範囲を設定したのですが、設定した可動範囲を越えて動いてしまいます。
モデルを下記のように記述しているのですが何が問題なのか分からず困っています。

ご教授よろしくお願いします。

[image]
[image]

以下にモデルの記述を示します。

format: ChoreonoidBody
formatVersion: 1.0
angleUnit: degree
name: Test

links:
  -
    name: BOX
    jointType: free
    centerOfMass: [ 0, 0, 0 ]
    mass: 5.0
    inertia: [
      0.1, 0,   0,
      0,   0.1, 0,
      0,   0,   0.5 ]
    elements:
      Shape:
        geometry:
          type: Box
          size: [ 0.5, 0.3, 0.085 ]
        appearance: 
          material:
            diffuseColor: [ 1, 0, 0 ]
            specularColor: [ 0.2, 0.8, 0.2 ]
            shinines: 0.6
  -
    name: BOX2
    translation: [ 0, 0, 0.0725]
    parent: BOX
    jointType: prismatic
    jointRange: [ -0.002, 0.002 ]
    jointAxis: x
    centerOfMass: [ 0, 0, 0 ]
    mass: 5.0
    inertia: [
      0.1, 0,   0,
      0,   0.1, 0,
      0,   0,   0.5 ]
    elements:
      Shape:
        geometry:
          type: Box
          size: [ 0.68, 0.135, 0.06 ]
        appearance: 
          material:
            diffuseColor: [ 1, 0, 1 ]
            specularColor: [ 0.2, 0.8, 0.2 ]
            shinines: 0.6

s-nakaoka, 2017-09-19:
jointRangeに設定した値がシミュレーション中に有効になるかどうか（境界で止まるかどうか）については、シミュレータアイテムのタイプ（≒物理エンジンのタイプ）によります。
恐らく標準のAISTSimulatorItemを使っておられるかと思いますが、そちらは現在のところこれが有効となりません。
確かODESimulatorItemなどでは有効となるよう実装していたように思います。
今は商用物理エンジンであるAGX Dynamicsの組み込みに力を入れておりまして、そちらについてもjointRangeが有効となる見込みです。

YudaiTogawa, 2017-09-20:
ご回答ありがとうございます。他の手法を検討してみます。

YudaiTogawa, 2017-09-21:
ODEシミュレータで試してみたのですが、可動範囲の設定がうまくいかないので再度質問させていただきます。
下図のようにODEsimulatorItemを使ってシミュレーションを行ったのですが可動範囲を越えて動く理由がわからず困っています。bodyファイルは上の質問に載せているものと変わらないです。
[image]

s-nakaoka, 2017-09-21:
今ODESimulatorItemの実装を見返してみたのですが、アイテムのプロパティとして「関節範囲拘束」というのがあって、これがデフォルトではfalseとなっていますが、これをtrueにすると関節範囲の拘束が効くようになるようです。（すみません、このプロパティがあることには気づいたのですが、まだうまく動くか確認は出来ていません・・・。）

YudaiTogawa, 2017-09-22:
ありがとうございます、間接範囲拘束を有効にしたところ、うまくいきました。ご教授ありがとうございました。

closed 2017-09-22 by YudaiTogawa