s-nakaoka/choreonoid #166: cannot compile a controller

piekah, 2017-09-19:
I'm still a student and just have an experience with C language for about 3 years.
I make a simple project and tried to compile a Controller as written on : file:///home/pieka/choreonoid/choreonoid-manual-2017-05-08/simulation/howto-implement-controller.html ; but I can't get it compile along with Choreonoid apps.
I tried the cmake option but there only this error message : 

CMake Error at CMakeLists.txt:2 (add_cnoid_simple_controller):
  Unknown CMake command "add_cnoid_simple_controller".
-- Configuring incomplete, errors occurred!
See also "/home/user/choreonoid/ext/ball/CMakeFiles/CMakeOutput.log".

Can anyone help me?

s-nakaoka, 2017-09-19:
Could you explain what did you do in your command-line when you tried to compile your controller?

piekah, 2017-09-20:
I just use make command and this is what I got. It seems like the controller(BallController) that I want to compile is not included.

Vostro-3558:~/choreonoid$ make
[  1%] Built target irrXML
[ 14%] Built target CnoidUtil
[ 15%] Built target CnoidPython
[ 17%] Built target PyUtil
[ 24%] Built target CnoidAISTCollisionDetector
[ 35%] Built target CnoidBody
[ 35%] Built target PyBody
[ 68%] Built target CnoidBase
[ 68%] Built target CnoidPyBase
[ 71%] Built target PyBase
[ 71%] Built target PyQtCore
[ 71%] Built target PyQtGui
[ 72%] Built target CnoidAssimpPlugin
[ 81%] Built target CnoidBodyPlugin
[ 82%] Built target PyBodyPlugin
[ 86%] Built target CnoidPoseSeqPlugin
[ 87%] Built target PyPoseSeqPlugin
[ 88%] Built target CnoidBalancerPlugin
[ 88%] Built target CnoidSimpleController
[ 89%] Built target CnoidSimpleControllerPlugin
[ 90%] Built target PySimpleControllerPlugin
[ 91%] Built target CnoidPythonPlugin
[ 91%] Built target PyPythonPlugin
[ 92%] Built target CnoidPythonSimScriptPlugin
[ 93%] Built target PyPythonSimScriptPlugin
[ 94%] Built target choreonoid
[ 94%] Built target SimpleVehicle
[ 94%] Built target PA10PickupController
[ 95%] Built target SampleSVController
[ 95%] Built target SR1LiftupController
[ 95%] Built target CameraSampleController
[ 96%] Built target SR1WalkPatternController
[ 96%] Built target SampleCrawlerController
[ 96%] Built target SR1WalkGraspController
[ 96%] Built target SampleCrawlerJoystickController
[ 97%] Built target TankJoystickController
[ 98%] Built target SR1MinimumController
[ 98%] Built target TankLightController
[ 98%] Built target 4WheelsJoystickController
[ 98%] Built target ConveyorController
[ 98%] Built target TankTutorial_TrackController
[ 98%] Built target TankTutorial_LightController
[ 98%] Built target TankTutorial_TurretController2
[ 99%] Built target TankTutorial_TurretController1
[100%] Built target TankTutorial_CameraController

s-nakaoka, 2017-09-20:
It seems that other SimpleController files are successfully compiled.
Where is your BallController? What kind of files are there?

piekah, 2017-09-20:
Here is the files and BallController.cpp in ext directory as stated below: 

Vostro-3558:~/choreonoid/ext/ball$ ls
BallController.cpp  CMakeCache.txt  CMakeFiles  CMakeLists.txt  ball.body

I had tried the "cmake .." command and what I got is only the error message on top.

s-nakaoka, 2017-09-20:
> I had tried the "cmake .." command and what I got is only the error message on top.

In this case, you only have to do cmake for the main Choreonoid source code.

piekah, 2017-09-21:
I did as what you said and I got this : 
Vostro-3558:~/choreonoid$ cmake .. 
CMake Error: The source directory "/home/user" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.

piekah, 2017-09-21:
Do you think I should uninstall the apps and install it back?

s-nakaoka, 2017-09-21:
Please try the following commands:

cd
cd choreonoid
make clean
rm CMakeCache.txt
rm ext/ball/CMakeCache.txt
cmake .
make

piekah, 2017-09-22:
I did as you said but all I got is this : 
Vostro-3558:~/choreonoid$ make
[  1%] Built target irrXML
[ 13%] Built target CnoidUtil
[ 14%] Built target CnoidPython
[ 16%] Built target PyUtil
[ 22%] Built target CnoidAISTCollisionDetector
[ 32%] Built target CnoidBody
[ 32%] Built target PyBody
[ 64%] Built target CnoidBase
[ 64%] Built target CnoidPyBase
[ 67%] Built target PyBase
[ 67%] Built target PyQtGui
[ 68%] Built target PyQtCore
[ 69%] Built target CnoidAssimpPlugin
[ 78%] Built target CnoidBodyPlugin
[ 79%] Built target PyBodyPlugin
[ 83%] Built target CnoidPoseSeqPlugin
[ 84%] Built target PyPoseSeqPlugin
[ 84%] Linking CXX shared library ../../lib/choreonoid-1.6/libCnoidBalancerPlugin.so
[ 85%] Built target CnoidBalancerPlugin
[ 86%] Built target CnoidSimpleController
[ 87%] Built target CnoidSimpleControllerPlugin
[ 88%] Built target PySimpleControllerPlugin
[ 89%] Built target CnoidPythonPlugin
[ 90%] Built target PyPythonPlugin
[ 91%] Built target CnoidPythonSimScriptPlugin
[ 91%] Built target PyPythonSimScriptPlugin
[ 92%] Built target choreonoid
[ 93%] Built target SimpleVehicle
[ 93%] Built target PA10PickupController
[ 93%] Built target SampleSVController
[ 94%] Built target SR1LiftupController
[ 95%] Built target CameraSampleController
[ 96%] Built target SR1WalkPatternController
[ 96%] Built target SampleCrawlerController
[ 97%] Built target SR1WalkGraspController
[ 98%] Built target SampleCrawlerJoystickController
[ 99%] Built target TankJoystickController
[ 99%] Built target SR1MinimumController
[ 99%] Built target TankLightController
[ 99%] Built target 4WheelsJoystickController
[100%] Built target ConveyorController

s-nakaoka, 2017-09-26:
I'm not sure why you cannot build your controller.
Perhaps are there still some CMakeCache.txt files in other directories such as choreonoid/ext? The cache files that are generated in a wrong way seem to make troubles.

piekah, 2017-09-26:
I tried to do it again with the same method after deleted the CMakeCache.txt files including inside the choreonoid/ext folder. Then I setup the "ADDITIONAL_EXT_DIRECTORIES" using this path :
/home/pieka/choreonoid/ext
and yet still, the BallController doesn't seem to be compiled.

s-nakaoka, 2017-09-26:
Why did you execute cmake at the ext or ball directories?
There is no such description in the manual, and I pointed out you don't have to do it.
> In this case, you only have to do cmake for the main Choreonoid source code.

piekah, 2017-09-26:
I did execute on the main source before I execute it at the ext or ball directories but the BallController didn't compiling.

s-nakaoka, 2017-09-26:
I mean you must not execute cmake at the ext or ball directories.

piekah, 2017-09-26:
Okay I got it. As for the path that I had setup, was it the right path?

s-nakaoka, 2017-09-26:
CMakeLists.txt contained in the sub directory of the ext directory is automatically detected when you execute cmake for the main Choreonoid source. In this case, you don't have to specify the directory to ADDITIONAL_EXT_DIRECTORIES.

piekah, 2017-09-26:
> CMakeLists.txt contained in the sub directory of the ext directory..

In my ext directory there are no CMakeLists.txt.

s-nakaoka, 2017-09-26:
My description refers to **a sub directory** of the ext directory.

s-nakaoka, 2017-09-26:
I think It is difficult to help you any more.
This is not a support center for you.
Please let me close this issue.

closed 2017-09-26 by s-nakaoka

s-nakaoka, 2017-09-26:
@piekah Please read the following document.
http://www.catb.org/~esr/faqs/smart-questions.html