s-nakaoka/choreonoid #170: 4輪サンプルモデルの車輪について

yukimorikane, 2017-10-03:
今回提供されている4輪モデルの車輪のサイズのみを変更したところ図のように前輪が内側に入り込みました。このような場合はコントローラモジュール、もしくはモデルファイルのどちらを変更すれば良いのでしょうか。
[image]

s-nakaoka, 2017-10-03:
サイズが変わった分ステアリングロッドの接続位置も変わってしまって、左前輪のデフォルトの方向が右側に切れ込んでしまったということでしょうか。
これを真っ直ぐにしたいということでしたら、モデルを直すのがよろしいかと思います。

yukimorikane, 2017-10-03:
ステアリングロッドの位置を変更することで改善されるということでしょうか？

s-nakaoka, 2017-10-03:
よく分かりません。あなたがモデルを変更したんですよね？私はその内容を知らないのに、どうやって回答すればよいのですか。

s-nakaoka, 2017-10-03:
piekahというアカウントの方もそうでしたが、ここをサポートデスクと思ってらっしゃるのかもしれませんが、チャットのように使うのはやめてください。質問するならしっかりと内容を整理して、他の方が分かるように書いて下さい。また自分でできることは調べたり試したりして、それでも解決できない部分を質問するようにしてください。ついた回答についてもよく読んでください。よく分からなければ、まず以下のようなページを読むようにしてください。
http://www.hyuki.com/writing/techask.html
http://www.catb.org/~esr/faqs/smart-questions.html

yukimorikane, 2017-10-03:
説明不足で申し訳ありません。具体的に変更した点は、前輪および後輪のgeometry Cylinder内のradiusを0.1→0.08に、heightに関しては変更していません。現在はこの点以外の数値を変更していません。

DEF 4wheelsVehicle Humanoid {
 
  humanoidBody [

    DEF BODY Joint {
      jointType "free"
      translation 0 0 0.1
      children [
        DEF BODY_LINK Segment {
          centerOfMass  0 0 0
          mass    10.0
          momentsOfInertia [ 0.217 0 0 0 0.083 0 0 0 0.283 ]
          children [
            Transform {
              translation 0 0 0
              children Shape {
                appearance DEF green Appearance {
                  material Material {
                    diffuseColor 0.0 0.6 0.0
                    ambientIntensity 0.3
                    specularColor 0.7 0.7 0.7
                    emissiveColor 0 0 0
                    shininess 0.25
                    transparency 0
                  }
                }
                 geometry Box {
                   size 0.3 0.5 0.1
                }
              }  # End of Shape node
            }
            Transform {
              translation   0 0.09 0.07
              children Shape {
                appearance USE green
                geometry Box { size 0.2 0.3 0.1 }
              }
            }
          ]
        }  # End BODY_LINK Segment

        DEF STEERING_RIGHT Joint {
          translation -0.15 -0.2 0
          jointType "rotate"
          jointAxis 0 0 1
          jointId  0
          children [
            DEF STEERING_RIGHT_LINK Segment {
              centerOfMass  0 0.025 0
              mass    0.5
              momentsOfInertia [ 0.0001 0 0 0 0.00003 0 0 0 0.0001 ]
              children [
                Transform {
                  translation 0 0.025 0
                  rotation 0 0 1 0
                  children Shape {
                    appearance Appearance {
                      material DEF black Material {
                        diffuseColor 0.01 0.01 0.01
                        ambientIntensity 0.01
                        specularColor 0.3 0.3 0.3
                        emissiveColor 0 0 0
                        shininess 0.01
                        transparency 0
                      }
                    }
                    geometry Cylinder{
                      radius 0.01
[... 154 lines cut ...]
                      height 0.04
                    }
                  }  # End of Shape node
                }
                Transform {
                  translation 0.175 0 0
                  rotation 0 0 1 1.570796327
                  children Shape {
                    appearance Appearance {
                      material USE black
                    }
                    geometry Cylinder{
                      radius 0.1　←←
                      height 0.04
                    }
                  }  # End of Shape node
                }
              ]
            } # segment FRONT_WHEEL_LINK
          ]
        }  # Joint REAR_WHEEL
      ]
    }
  ]
}

s-nakaoka, 2017-10-03:
なるほど、タイヤの半径を変えたのですね。

モデルファイルを提示していただきましたが、半径0.1を0.08に変えたということでどこが0.08になっているのか探したのですが見つかりませんでした。よく見たところ、これは0.08に値を変えたものではなく、元の0.1のものの変更すべき箇所に矢印の文字をつけたものだったのですね。これは少し分かりにくいです。このままではChoreonoidで読み込めませんし。一般的には、変更してうまく動かなかったファイルを提示した方がよいかと思います。（全て提示すると大きくなる場合は、変更した箇所だけでもよいかと思います。）

意図は分かりましたので、こちらで0.08に変えて読み込んでみました。そこで分かったのですが、モデルファイルを読み込んだ時点でおかしくなるわけではないようですね。車輪についても真っ直ぐに表示されています。

恐らくこれをシミュレーションで動かすと添付の図のように前輪がうまくコントロールできなくなるということかと思います。ただ、何のコントローラで動かしたのかが不明です。

まずは、上記のようなことを、こちらがいろいろと試したり予想したり、ひとつひとつ聞いたりしなくても済むように、最初の質問の際になるべく全て明らかにしてください。そうすると、回答もつきやすいのではないかと思います。先ほど挙げた質問心得のページにもそのようなことが書いてあるかと思いますが、そちらは読みましたか？

s-nakaoka, 2017-10-03:
回答の続きですが、まずそちらで使っているVRMLファイルは古いです。今は多くのサンプルモデルをVRMLではなく.bodyの形式で書くようにしていて、最新版でもそうなっています。この４輪モデルについてはshare/model/misc/FourWheelCar.body というファイルがありますので、そちらを使うようにしてください。
また、これのサンプルもあります。そちらはshare/project/FourWheelCarJoystick.cnoid というプロジェクトファイルです。これを実行するとジョイスティックで車を操作できます。
そこでこのサンプルでタイヤの半径を0.08に変えて試してみたのですが、それでも正しくシミュレーション出来ました。恐らくそちらでお使いのコントローラがモデルとあっていないか、どこかおかしなところがあるのだと思います。まずはChoreonoidをアップデートして、上記のサンプルを試して、そちらを改良することでやりたいことをできないか試してみてもらえますでしょうか。

yukimorikane, 2017-10-05:
返答ありがとうございます。指摘された点について試してみたいと思います。
質問の仕方についてはこれから注意していきます。

yukimorikane, 2017-10-16:
アップデートを行いサンプルのタイヤの半径を変更したところ無事動作の確認をすることができました。ありがとうございます。
しかし、share/model/tank/tank.bodyをChoreonoidで読み込んだところ、メッセージの部分に
"tank.dae"のファイル形式はシーンローダでサポートされていません。
という表示が出ました。これは私の行ったアップデートに何らかの問題があるということでしょうか。

s-nakaoka, 2017-10-16:
拡張子daeのモデルはCOLLADA形式となりますが、この形式を読みこむためにはAssimpPluginが必要です。

開発版ではUbuntu14.04、16.04に関してはデフォルトでONになるようにしてありますが、以前からお使いの場合、依存パッケージがインストールされていなかったり、以前のCMakeの設定が残っていたりして、そのままではONになりません。

まずmisc/script/insutall-requisities-ubuntu-16.04.sh などのパッケージインストールスクリプトを再度実行してください。その後、CMakeでBUILD_ASSIMP_PLUGINをONにしてビルドしてみてください。

s-nakaoka, 2017-11-07:
@yukimorikane さん。
こちらの件はその後どうなりましたでしょうか、Issueを閉じてもよろしいでしょうか？
Issueというのは本来不具合や改善要望などについて報告・議論し、議論が終わったら閉じるという性質のものですし、一般的な話としても質問に対して回答すればその結果が気になるものですから、そのようなことを踏まえた対応をお願いします。繰り返しになって恐縮ですが、以下のページも参考にしてください。
http://www.hyuki.com/writing/techask.html

yukimorikane, 2017-11-08:
返信をしなかったことは王仕訳ございませんでした。
言われたとおりのことを実行した結果うまくいくことができました。

s-nakaoka, 2017-11-08:
お知らせいただきありがとうございます。
うまくいったようでよかったです。
ではこのIssueは閉じておきます。

closed 2017-11-08 by s-nakaoka