s-nakaoka/choreonoid #173: VisionSensorだけのBodyRTCがrtshellで扱えない（omniORB.MARSHAL_InvalidEnumValue）

y-masutani, 2017-11-01:
環境に固定されたカメラを模擬するために，VisionSensorだけを含むVRMLファイルを作り，BodyRTCを使って外部のRTCと接続して使おうとしています．RT System Editorで使う場合は問題ないのですが，rtshellで扱うことができず困っています．
```
rtls /localhost/Camera.rtc
rtls: CORBA.MARSHAL(omniORB.MARSHAL_InvalidEnumValue, CORBA.COMPLETED_YES)
```
というエラーになります．rtlsだけでなく，rtconなどでも同じです．シミュレーションを一度実行すると，このエラーはなくなります．

検索をしたところ，https://github.com/s-nakaoka/choreonoid/issues/104 がいかにも関係していそうですが，具体的にどのようにすればいいかわかりません．

WindowsでChoreonod-1.5.0をVS2012で自分でビルドして使っています．

s-nakaoka, 2017-11-07:
すみません、Choreonoidの古いバージョンでもあり、当方では今のところrtshellを使っていないので、対応が難しいです。RTM開発者の方で何か心当たりのある方がいらっしゃるとよいのですが・・・。まずは最新版でUbuntu環境からになりますが、rtshellとの連携ができるかについて確認してみようと思います。

s-nakaoka, 2017-11-20:
OpenRTMのMLで進展がありましたので、こちらは閉じておきます。

closed 2017-11-20 by s-nakaoka