s-nakaoka/choreonoid #174: YAML形式のモデルファイルについて

yukimorikane, 2017-11-07:
choreonoidのマニュアルにあるモデルファイル / Blenderを使用したモデルファイルの読み込み / YAML形式でのモデルファイルの読み込みについて質問があります。Blenderで作成したものをVisualを用いて表示させることは理解できました。しかし、Collisionのelements以下にあるtranslation、geometryの値の設定の仕方が分かりません。
自分で試みた方法はCollisionの下のelementsの1行を消去することでshapeで定義したものがchoreonoid上に表示されるということが分かりました。そこでこの可視化されたCollisionのShapeで定義されたモデルをVisualで読み込んだモデルの適切な位置にtranslationで設定し、再度elementsの1行を加えなおすことで作成しました。しかしこの方法では、目視で判断しているため少なからず位置や形に誤差が起きてしまいます。実際にはどうやるべきなのかを教えていただけないでしょうか。

anazawa-t, 2017-11-07:
Collision以下のtranslation、geometryの設定についてですが、Blenderでモデルを作成する際に、translationやsizeの設定を行っていると思います。基本的にはその値を元にChoreonoid側でも設定を行っております。
例えば、Blenderでsize（1.0, 1.0, 1.0）のBoxの中心が座標の原点となるようにBoxを配置するとします。
Choreonoid側のCollisionの設定で2つのBoxを用いて表現するには、まず立方体のBoxを半分にすると、半分にした直方体の中心がどこで、サイズいくつか分かると思います。
ただし、斜めになっている箇所等もあるため、上記のような当てはめ方ができない場合もあるので、その場合はChoreonoid上のシーンビュー上でマウスカーソルを当てると当てた箇所のグローバル座標が分かりますので、その値を元に目視で微調整を行い、当てはめる必要があります。

yukimorikane, 2017-11-07:
返信ありがとうございます。
その方法を試してみたいと思います。

closed 2017-11-20 by s-nakaoka