s-nakaoka/choreonoid #175: コントローラのコンパイルについて

HiroshiIzumitani, 2017-11-08:
府大高専の和泉谷です。

コントローラのコンパイルを行うために “CMakeLists.txt”を作成し、”cd”で移動したあと”cmake .”を行ったのですが次のようなエラーが発生し、

CMake Error at CMakeLists.txt:1 (add_cnoid_simple_controller):
  Unknown CMake command "add_cnoid_simple_controller".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.5)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/nakatani10/Downloads/choreonoid-master/sample/SimpleController/CMakeFiles/CMakeOutput.log".

”make”を実行すると

make: *** ターゲットが指定されておらず, makefile も見つかりません.  中止.

というエラーが出ました。

この場合、どのように対処すればmakefileを見つけ出すかコンパイルを完了させることができるでしょうか。

返信おねがいします。

s-nakaoka, 2017-11-08:
Tankチュートリアル
http://choreonoid.org/ja/manuals/latest/simulation/tank-tutorial/index.html
の「コントローラのコンパイル方法」
http://choreonoid.org/ja/manuals/latest/simulation/tank-tutorial/step2.html#tank-tutorial-step2-compile
を参考にしてください。

add_cnoid_simple_controllerでコンパイルを行う場合は、上記ページに書かれている「2. Choreonoid本体と一緒にコンパイルを行う」に該当します。この場合、コントローラ用に書かれたCMakeLists.txtに対してcmakeを行うのではなく、Choreonoid本体のCMakeLists.txtからコントローラのCMakeLists.txtが読み込まれるようにした上で、Choreonoid本体のcmakeを実行するということになります。その上で、Choreonoid本体のmakeを行うと同時にコントローラも生成されるということになります。

s-nakaoka, 2017-11-09:
補足です。和泉さんがやられたようにいったん適切でない場所でcmakeを実行してしまうと、cmakeの設定のキャッシュのファイルが残ってしまい、これが悪さをして、その後正しい方法でcmakeやmakeをやろうとしても、うまくいかなくなることがあるようです。この問題への対処方法としては、cmakeが生成するCMakeCache.txtというファイルを消して下さい。Choreonoid本体のビルドディレクトリに生成されたものと、コントローラ用に間違ってcmakeしてしまって生成されたものを、両方消しておくとよいかと思います。よく分かりませんが、CMakeFilesというディレクトリも消しておいたほうがいいかもしれません。その上で、Choreonoid本体*のみ*に対して、cmakeを実行しなおして、makeします。

HiroshiIzumitani, 2017-11-09:
和泉谷です。

さきほど、cmakeが生成するCMakeCache.txtというファイルを消して、Choreonoid本体のビルドディレクトリに生成されたものと、コントローラ用に間違ってcmakeしてしまって生成されたものを、両方消しました。また、MakeFilesというディレクトリも消しました。

その後、Cmakeを行ったのですが、

nakatani10@nakatani10-H170M-S01:~/Downloads/choreonoid-master$ cmake .-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
--   regex
--   thread
--   iostreams
--   date_time
--   chrono
--   atomic
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   chrono
--   system
CMake Error at src/Body/CMakeLists.txt:180 (add_library):
  add_library cannot create target "TankJoystickController" because another
  target with the same name already exists.  The existing target is a shared
  library created in source directory
  "/home/nakatani10/Downloads/choreonoid-master/sample/SimpleController".
  See documentation for policy CMP0002 for more details[.](url)
Call Stack (most recent call first):
  sample/SimpleController _org/CMakeLists.txt:40 (add_cnoid_simple_controller)

-- Configuring incomplete, errors occurred!
See also "/home/nakatani10/Downloads/choreonoid-master/CMakeFiles/CMakeOutput.log".
See also "/home/nakatani10/Downloads/choreonoid-master/CMakeFiles/CMakeError.log".

というようにエラーが出てきました。これはいまどんのような問題が発生しているのでしょうか。また、「Choreonoid本体のCMakeLists.txtからコントローラのCMakeLists.txtが読み込まれるようにした上で、Choreonoid本体のcmakeを実行する」「Choreonoid本体のmakeを行うと同時にコントローラも生成される」とは具体的にどのような手順で行うのでしょうか。

返信おねがいします。

"/home/nakatani10/Downloads/choreonoid-master/CMakeFiles/CMakeOutput.log" "/home/nakatani10/Downloads/choreonoid-master/CMakeFiles/CMakeError.log"も添付します。
[CMakeError.log](https://github.com/s-nakaoka/choreonoid/files/1456779/CMakeError.log)
[CMakeOutput.log](https://github.com/s-nakaoka/choreonoid/files/1456780/CMakeOutput.log)

s-nakaoka, 2017-11-09:
エラーメッセージでは、"TankJoystickController"というターゲットが他にも存在する、とありますね。Choreonoid本体と一括してコンパイルする都合上、Choreonoid本体に含まれているサンプルの識別名とも重ならないようにする必要があります。新たに作成したCMakeListst.txtでこの名前を使っているかと思いますので、名前を変えるようにしてください。

> 「Choreonoid本体のCMakeLists.txtからコントローラのCMakeLists.txtが読み込まれるようにした上で、Choreonoid本体のcmakeを実行する」「Choreonoid本体のmakeを行うと同時にコントローラも生成される」とは具体的にどのような手順で行うのでしょうか。

上でお知らせしたTankチュートリアルの「コントローラのコンパイル方法」
http://choreonoid.org/ja/manuals/latest/simulation/tank-tutorial/step2.html#tank-tutorial-step2-compile
に書いてありますので、それを読んでください。

HiroshiIzumitani, 2017-11-16:
わかりました。

返信遅れました。丁寧にアドバイスありがとうございました。何かあればまたお願いします。

closed 2017-11-16 by s-nakaoka