s-nakaoka/choreonoid #176: Pythonのスクリプトの読み込ませ方

y-masutani, 2017-11-08:
https://github.com/s-nakaoka/choreonoid/issues/64 を参考に以下のような摩擦を設定するPythonのスクリプトを作りました．
```
from cnoid.Base import *
from cnoid.BodyPlugin import *

ric30 = Item.find("RIC30").body()
floorLink = Item.find("Floor").body().rootLink()
simulator = Item.find("AISTSimulator")

simulator.setFriction(ric30.link("L_ANKLE_R"), floorLink, 0.0, 0.0)
```
Choreonoidを実行して，Pythonコンソールビューに上記のスクリプトを流し込むと所望の動作をします．同じことをメニューのファイル→読み込み→Pythonスクリプトでやりたいのですが，うまくいきません．メッセージビューには，
```
Pythonスクリプト "C:\Users\（中略）\RIC30-friction.py" を読み込み中
 -> 完了!
```
と表示されていますが，設定が反映されません．

また，起動時に--pythonオプションでスクリプトファイルを指定した場合は，設定は反映されますが，起動時にメッセージビューに以下のように表示されます．
```
Pythonスクリプト ".\RIC30-friction.py" を実行中...
.\RIC30-friction.py:1: RuntimeWarning: to-Python converter for class cnoid::ref_ptr<class cnoid::Item> already registered; second conversion method ignored.
  from cnoid.Base import *
.\RIC30-friction.py:1: RuntimeWarning: to-Python converter for class cnoid::ref_ptr<class cnoid::RootItem> already registered; second conversion method ignored.
  from cnoid.Base import *
.\RIC30-friction.py:2: RuntimeWarning: to-Python converter for class cnoid::ref_ptr<class cnoid::Body> already registered; second conversion method ignored.
  from cnoid.BodyPlugin import *
.\RIC30-friction.py:2: RuntimeWarning: to-Python converter for class cnoid::ref_ptr<class cnoid::Link> already registered; second conversion method ignored.
  from cnoid.BodyPlugin import *
.\RIC30-friction.py:2: RuntimeWarning: to-Python converter for class cnoid::ref_ptr<class cnoid::SimulatorItem> already registered; second conversion method ignored.
  from cnoid.BodyPlugin import *
スクリプトが終了しました。
```
ファイル→読み込み→Pythonスクリプトの正しい使い方を教えてください．

Windows上のChoreonoid 1.5（インストーラでインストールしたもの）で試しています．

hattorishizuko, 2017-11-09:
マニュアルの不備が多くて、大変申し訳ございません。

シミュレーションの設定を行うスクリプトは、
ファイル->読み込む->シミュレーション用Pythonスクリプト　で読み込んください。
アイテムビューにアイテムが作成されますので、シミュレーションアイテムの子アイテムとして配置し、チェックボックスにチェックを入れます。
（これで、シミュレーション実行時に自動で実行されるようになります。Pythonスクリプトとして読み込んだ場合は、スクリプト実行ボタンを押して手動で実行するような場合に使用します。）

実行のタイミングは、プロパティビューで設定します。
シミュレーション用Pythonスクリプトのプロパティを表示し、タイミングの項目で初期化前を選択してください。

エラーメッセージに関しましては、調査が必要なのでお時間頂きたいと思います。

y-masutani, 2017-11-09:
ありがとうございます．うまくいきました．
いろいろ試行錯誤していたのですが，「タイミングの項目で初期化前」に気が付きませんでした．

closed 2017-11-09 by y-masutani