s-nakaoka/choreonoid #178: GLVisionSimulatorItem not working?

yosuke, 2017-11-15:
GLVisionSimulatorItemを使った視覚シミュレーションをしようとしているのですが、どうにも動かず、ここで質問させてください。

構成としては、choreonoidをVisionSensorサンプル付きでコンパイルして、以下のようにTankモデルを構成しています。

[image]

メッセージからVisionSensorビューとGLVisionSImulatorItemの両者からモデルのカメラを検出できていることを確認できるのですが、その後のGLVisionSImulatorItem上でのレンダリングが動いていないようで、どうにも画像が取り出せません。

VisionSensorプラグインを改変した形のプラグインが作りたかったので、camera->sigStateChanged()でログを出力させてみたのですが、シミュレーション開始時の１回しかシグナルが発行されていない状態です（その時点での画像サイズは０）。試しにと思い、bodyItem->sigKinematicStateChanged()のイベントで定期的にカメラの画像を見てみたのですが、常にサイズ０で更新されていないような状態です。

環境としては、ubuntu 16.04、複数の環境で試しており、nvidiaのプロプライエタリドライバ、virtualbox上でのmesaを使ったソフトウェアレンダリング、masterブランチとrelease-1.5ブランチ、USE_QT5オプションの有効化を試してもいずれもだめです。

各環境でのglxinfoの出力は以下のようになっています。

virtualbox(mesa-gl)上
```
$ glxinfo | grep OpenGL
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 4.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 17.0.7
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 17.0.7
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 17.0.7
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
```

nvidiaプロプライエタリドライバ
```
$ glxinfo | grep OpenGL
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1050 Ti/PCIe/SSE2
OpenGL core profile version string: 4.5.0 NVIDIA 387.12
OpenGL core profile shading language version string: 4.50 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6.0 NVIDIA 387.12
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL extensions:
```

プロジェクトの設定の問題のような単純な原因の気もしないでもないのですが、原因に心当たりありましたら教えてください。

s-nakaoka, 2017-11-16:
> プロジェクトの設定の問題のような単純な原因の気もしないでもないのですが、原因に心当たりありましたら教えてください。

少し心当たりのある点として、この例では、GLVisionSimulatorItemの「ビジョンデータの記録」プロパティと、それと紐付けられたシミュレータアイテムの「デバイス状態の記録」が両方ともtrueになっている必要があるかもしれません。もしどちらかがfalseになっていましたら、trueにして試していただけますでしょうか。

これでうまく行くかは分かりませんが、補足としてこれらのプロパティの意味について説明します。

これらのプロパティについては、ログ記録と関連します。まず、「デバイス状態の記録」については、シミュレーションの内部処理で更新されたデバイスの状態について、GUI側のモデル（BodyItem)に伝えるかどうかを決定します。「記録モード」が「オフ」のときは、BodyItemに直接伝えますし、「記録モード」がそれ以外のときは、ログを介して伝えます。

さらに、カメラやレンジセンサの画像データの部分については、GLVisionSimulatorItemの「ビジョンデータの記録」がtrueだと、同様にGUI側のモデル(BodyItem)に伝わるようになります。

これらをfalseにしたとしても、シミュレーションの内部では更新されており、それはコントローラには伝わりますので、単にシミュレーションを行う場合でもあればこれでも問題ありません。

ただ、GUI側でBodyItemのシグナルを通してGUI側にデータを表示したいといったことになりますと、これらのプロパティをtrueにしておく必要があります。「ビジョンデータの記録」が別のプロパティになっているのは、ビジョンデータは一般的にサイズが大きくなりますので、それをログとして毎フレーム全て記録しておくことになると、メモリをかなり消費することになるので、この部分は別途設定できるようにした次第です。

一般的にはセンサのデータはコントローラ側で処理できればそれでよいことも多いので、このようなプロパティを導入しています。

VisionSensorサンプルはコントローラを介さずにGUI側に直接伝えて表示するというものなので、上記のプロパティをtrueにしてあります。今回の件でもBodyItemのオブジェクトからシグナルを検出して、ということだったので、同様のことが当てはまるかと思います。

ただしこれらのプロパティがtrueになっていたとしても更新されないなら、また別の問題になりますね・・・。その場合はまたお知らせ下さい。

yosuke, 2017-11-16:
シミュレータアイテムもビジョンシミュレータアイテムもデフォルト設定のまま使っていたのですが、「ビジョンデータの記録」がfalseになっており、それをtrueにしたところ無事に動き始めました。ありがとうございました。

ビジョンシミュレータアイテムのソースを見て、レンダリングしてsetImageした後にnotifyが出ていないなと思っていたのですが、その下にあるdeviceのほうでオプションをif条件にしてnofityがかかっているのが見つかりました。

ただ、やや腑に落ちないのは、記録のオプションがfalseの場合でも、レンダリングは行われてカメラのバッファへのsetImageも行われているはずだと思います。`camera->sigStateChanged()`のイベントが発行されないのは理解ができたのですが、`bodyItem->sigKinematicStateChanged()`のイベントを拾ってカメラのバッファを覗き込めば内容が見えても良さそうなものなのにな、、、とも思います。

オプション設定時に`camera->setImageStateClonable(true)`としているのも関係があるのかなとも思うのですが、もしお時間ありましたら教えてください。

とりあえずは省メモリなどを考えなくて良い状況なので、記録のオプションをtrueにしてこちらの作業は問題ありません。

s-nakaoka, 2017-11-16:
> ただ、やや腑に落ちないのは、記録のオプションがfalseの場合でも、レンダリングは行われてカメラのバッファへのsetImageも行われているはずだと思います。camera->sigStateChanged()のイベントが発行されないのは理解ができたのですが、bodyItem->sigKinematicStateChanged()のイベントを拾ってカメラのバッファを覗き込めば内容が見えても良さそうなものなのにな、、、とも思います。

このあたり少し分かりにくいかもしれませんが、そこで対象となっているのは、元のBodyItemのbodyオブジェクトからコピーされたもので、シミュレーション用スレッドで使われているものであり、BodyItemが所有するbodyオブジェクトとは異なるものです。シミュレーションはシミュレーション専用のスレッドで処理されるので、GUI（メインスレッド）側で管理しているBodyItem、bodyオブジェクトとは別のオブジェクトが必要になるというわけです。ただ表示はGUI(メインスレッド）側で行いますので、その間はスレッド競合をさけるかたちで、ログデータなどを介してシミュレーション用スレッドとメインスレッドとの間でデータをやり取りしています。カメラのバッファについても、元々のオブジェクトが異なっているので、シミュレーション用スレッドで使われているオブジェクトからGUIスレッドで使われているオブジェクトにコピーしない限り、GUI側での更新はされません。

s-nakaoka, 2017-11-16:
ちなみに先ほど改良したGLVisionSimulatorItemのコードをアップしておきました。この改良で広角(180度以上、360度も可能）のレンジセンサが使えるようになりました。ベロダインセンサもシミュレートできます。
サンプルとして、TankVisionSensors.cnoidというものを用意しています。デフォルトでベロダインセンサのシミュレーション結果が表示されます。

このサンプルはこれまでBUILD_VISION_SENSOR_SAMPLEをONにするとビルドされていたサンプルの後継となるものです。そちらはプラグインでコーディングしてセンサデータの可視化を行っていましたが、これを行う機能が標準機能（SensorVisualizerItem)となり、それを使うことでプラグインなしで可視化できるようになりました。これにともない上記オプションと対応するsample/VisionSensor は廃止しました。

yosuke, 2017-11-17:
情報ありがとうございます。フレームワーク的な設計の概念の部分なので理解するのにもう少し時間が必要そうですが、考えながらいじってみます。

ざっとコードを見たところ、SimulatorItemにfindSimulationBody()を見つけたのですが、このクラスがcontrollerを管理しており、controller側にはこのクラスのインスタンスが見えるのでセンサが触れるのだろうな、と理解しています。

GUI側からfindSimulationBody経由でbody本体を手に入れれば画像を取り出せなくもなさそうですが、設計思想にはあわなさそう＆せっかくスレッドセーフに設計しているのを崩しそう、ですね。
プラグイン内でコントローラを作って、そのコントローラをSimulationBodyに持ってもらって、ビジョンの処理はそのコントローラで動かしつつ、処理結果をGUIに送る部分はChoreonoidと同様につなぐ、というのが正しいやり方なのかなとぼんやりと考えています。

ちなみになのですが、「スレッド競合をさけるかたちで、ログデータなどを介してシミュレーション用スレッドとメインスレッドとの間でデータをやり取り」がコードとして理解できる部分がありましたら教えていただくことは可能でしょうか？

s-nakaoka, 2017-11-17:
> ちなみになのですが、「スレッド競合をさけるかたちで、ログデータなどを介してシミュレーション用スレッドとメインスレッドとの間でデータをやり取り」がコードとして理解できる部分がありましたら教えていただくことは可能でしょうか？

SimulatorItem.cpp の SimulatorItemImpl::flushResults 関数からこの処理を行っています。
シミュレーションスレッド側では、結果をバッファに一時的に記録しています。そしてflushResults関数内で、バッファ内のデータをGUIスレッド側のログデータ、もしくはBodyItem内のBodyに直接、コピーをします。バッファにアクセスする際にはmutexによって排他制御をかけていて、シミュレーションスレッドとGUIスレッドで競合しないようにしています。

closed 2017-12-06 by s-nakaoka