s-nakaoka/choreonoid #179: プラグイン実装時に出たエラーの意味について

HiroshiIzumitani, 2017-11-21:
こんにちは。府大高専の和泉谷です。

choreonoidに”JVRCPlugin”というプラグインを実装しようと思い、cmake,ccmakeを実行したあと、makeを実行したのですが、

nakatani10@nakatani10-H170M-S01:~/Downloads/choreonoid-master2$ cmake .
-- Boost version: 1.58.0
-- Found the following Boost libraries:
-- system
-- filesystem
-- program_options
-- regex
-- thread
-- iostreams
-- date_time
-- chrono
-- atomic
-- Boost version: 1.58.0
-- Found the following Boost libraries:
-- chrono
-- system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nakatani10/Downloads/choreonoid-master2
nakatani10@nakatani10-H170M-S01:~/Downloads/choreonoid-master2$ ccmake .

nakatani10@nakatani10-H170M-S01:~/Downloads/choreonoid-master2$ make
[ 13%] Built target CnoidUtil
[ 14%] Built target CnoidPyUtil
[ 17%] Built target PyUtil
[ 24%] Built target CnoidAISTCollisionDetector
[ 36%] Built target CnoidBody
[ 37%] Built target PyBody
[ 70%] Built target CnoidBase
[ 70%] Built target CnoidPyBase
[ 72%] Built target PyBase
[ 72%] Built target PyQtCore
[ 72%] Built target PyQtGui
[ 73%] Built target CnoidAssimpPlugin
[ 82%] Built target CnoidBodyPlugin
[ 83%] Built target PyBodyPlugin
[ 87%] Built target CnoidPoseSeqPlugin
[ 87%] Built target PyPoseSeqPlugin
[ 88%] Built target CnoidBalancerPlugin
[ 88%] Building CXX object src/JVRCPlugin/CMakeFiles/CnoidJVRCPlugin.dir/JVRCScoreView.cpp.o
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp: In constructor ‘cnoid::JVRCScoreViewImpl::JVRCScoreViewImpl(cnoid::JVRCScoreView*)’:
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp:232:13: error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(noColumn, QHeaderView::ResizeToContents);
^
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp:235:13: error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(taskColumn, QHeaderView::ResizeToContents);
^
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp:239:13: error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(eventColumn, QHeaderView::ResizeToContents);
^
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp:242:13: error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(autoTimeColumn, QHeaderView::ResizeToContents);
^
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp:245:13: error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(adoptDetectedTimeButtonColumn, QHeaderView::ResizeTo
^
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp:248:13: error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(manualTimeColumn, QHeaderView::ResizeToContents);
^
/home/nakatani10/Downloads/choreonoid-master2/src/JVRCPlugin/JVRCScoreView.cpp:251:13: error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(pointColumn, QHeaderView::Stretch);
^
src/JVRCPlugin/CMakeFiles/CnoidJVRCPlugin.dir/build.make:134: ターゲット 'src/JVRCPlugin/CMakeFiles/CnoidJVRCPlugin.dir/JVRCScoreView.cpp.o' のレシピで失敗しました
make[2]: *** [src/JVRCPlugin/CMakeFiles/CnoidJVRCPlugin.dir/JVRCScoreView.cpp.o] エラー 1
CMakeFiles/Makefile2:1492: ターゲット 'src/JVRCPlugin/CMakeFiles/CnoidJVRCPlugin.dir/all' のレシピで失敗しました
make[1]: *** [src/JVRCPlugin/CMakeFiles/CnoidJVRCPlugin.dir/all] エラー 2
Makefile:149: ターゲット 'all' のレシピで失敗しました
make: *** [all] エラー 2

と出てきて実装できませんでした。

そこで、
・　”error: ‘class QHeaderView’ has no member named ‘setResizeMode’
header->setResizeMode(autoTimeColumn, QHeaderView::ResizeToContents);”　の意味とどのような状態か

・このエラーの改善方法

を教えていただけないでしょうか。

よろしくお願いします。

hattorishizuko, 2017-11-21:
'class QHeaderView'には 'setResizeMode'という名前のメンバーがありません
と出力されている通りだと思います。

setResizeModeは、Qt5で廃止されているようです。下記をご覧ください。
http://doc.qt.io/qt-5/qheaderview-obsolete.html

HiroshiIzumitani, 2017-11-21:
'class QHeaderView'には 'setResizeMode'という名前のメンバーがありません
setResizeModeは、Qt5で廃止されているようです。
とありますがそれではこのJVRCScoreView.cpp内にあるこちらの”setResizeMode”は使用できないのでしょうか？

    recordTable.setHorizontalHeaderItem(noColumn, new QTableWidgetItem("No."));
    header->setResizeMode(noColumn, QHeaderView::ResizeToContents);

    recordTable.setHorizontalHeaderItem(taskColumn, new QTableWidgetItem("Task"));
    header->setResizeMode(taskColumn, QHeaderView::ResizeToContents);
    recordTable.setColumnHidden(taskColumn, true);
    
    recordTable.setHorizontalHeaderItem(eventColumn, new QTableWidgetItem("Event"));
    header->setResizeMode(eventColumn, QHeaderView::ResizeToContents);

    recordTable.setHorizontalHeaderItem(autoTimeColumn, new QTableWidgetItem("Detected Time"));
    header->setResizeMode(autoTimeColumn, QHeaderView::ResizeToContents);

    recordTable.setHorizontalHeaderItem(adoptDetectedTimeButtonColumn, new QTableWidgetItem(""));
    header->setResizeMode(adoptDetectedTimeButtonColumn, QHeaderView::ResizeToContents);
    
    recordTable.setHorizontalHeaderItem(manualTimeColumn, new QTableWidgetItem("Judged Time"));
    header->setResizeMode(manualTimeColumn, QHeaderView::ResizeToContents);

    recordTable.setHorizontalHeaderItem(pointColumn, new QTableWidgetItem("Point"));
    header->setResizeMode(pointColumn, QHeaderView::Stretch);

また使用できない場合どのようにすればよろしいでしょうか。
返信おねがいします。

s-nakaoka, 2017-11-21:
すみませんが質問の意図が分かりかねます。
「ChoreonoidにJVRCPluginを実装する」というのはどのような意味でしょうか？JVRCPluginを使えるようにしたい、ということでしょうか。それともこのプラグインを参考にご自分のプラグインを実装したいということでしょうか。

JVRCPluginは、2年前にJVRCという競技会を行った際に、この競技会で必要な機能を実装したものです。競技会は２年前のことでして、現在このプラグインはサポートされていません。当時のChoreonoidのバージョンを対象としているため、最新のChoreonoidバージョンに対してはそのままではビルドできないこともありまして、その点はご了承いただければと思います。

ただ、現在このプラグインをビルドして使う必要性は、一般的にはあまりないのではないかと思います。そこのところの意図がよく分からず、また質問をいただいたのはコンパイルエラーに関してでしたので、先のような回答になるかと思います。

HiroshiIzumitani, 2017-11-22:
先ほどは失礼しました。

JVRCPluginを使えるようにしたいのですが、現在このプラグインはサポートされていなく使えないということでよろしいでしょうか。

お願いします。

s-nakaoka, 2017-11-24:
はい、新しい環境で使う場合、現状のコードのままではコンパイルできないということのようですね。

HiroshiIzumitani, 2017-11-24:
わかりました。ありがとうございました。

closed 2017-12-06 by s-nakaoka