s-nakaoka/choreonoid #182: ゲームパッドを2台使う方法

YudaiTogawa, 2017-12-05:
ロボットモデルを操作するにあたり、ゲームパッドを2台使いたいと考えています。
そこで、タンクチュートリアルを参照したところ

「2台目以降のゲームパッドを使いたい場合はjs1、js2といったデバイスを使いますが、その場合はソースコードの変更が必要になりますのでご注意下さい。」

と記載されていました。
ソースコードの変更とは、どこを変更すればよいのでしょうか？

ご教授ください。
よろしくお願いします。

s-nakaoka, 2017-12-06:
Joystickクラスのコンストラクタにデバイスファイルのパスを指定します。
Ubuntuでは /dev/input/js0 といったパスになります。（最後の0のところが変わる。）
チュートリアルではJoystickをメンバ変数として定義したので、そのコンストラクタはコントローラのコンストラクタの初期化リストで記述します。

例えばTurretController2でしたら、

    TurretController2()
      : joystick("/dev/input/js1")
    {
    
    }

みたく書きます。

s-nakaoka, 2017-12-06:
ちなみにこういうときの調べ方について。
このチュートリアルではジョイスティックからの読み出しにJoystickクラスを使っていますので、読み込み先を変えるのもこのクラスの何らかの関数でできるんじゃないかなぁと目星をつけます。

次に

    #include <cnoid/Joystick>

という記述があるので、ジョイスティッククラスの定義はこのヘッダにあるだろうなぁと目星をつけます。
ソース内でこのファイルを探すと、include/cnoid/Joystick というファイルがあることが分かります。そこでこのファイルの中身をみると、

    #include "src/Util/Joystick.h"

と書いてあるので、実際の定義はこのファイルの中に書いてあることが分かります。
そこで src/Util/Joystick.h を読むと、

    Joystick(const char* device);

というコンストラクタがあることが分かります。デバイスを指定していそうな関数はこれくらいなので、まずはこれを使ってみよう、となります。

s-nakaoka, 2017-12-20:
本件結果はいかがでしょうか？チケットを閉じてもよろしいですか？ @YudaiTogawa

closed 2017-12-30 by s-nakaoka