s-nakaoka/choreonoid #184: Polar Positioning

Nbot07, 2017-12-16:
Is there a way to specify the locations of model attributes using polar coordinates in  Choreonoid?

s-nakaoka, 2017-12-20:
Body/Link View can be used to specify the global position and orientation of a model in polar-coordinate like values.
http://choreonoid.org/en/manuals/latest/handling-models/pose-editing.html#body-link-view

Here, x, y correspond to the position on the ground, and y(Yaw, the right bottom entry) corresponds to the orientation around yaw axis. The values correspond to (x, y, theta) like expression.

s-nakaoka, 2017-12-20:
Sorry, (x, y, theta) expression is not the polar-coordinate. You probably mean (r, theta), where r is the distance from the origin, and theta is the angle around the Z-axis. Currently there is no such an interface in Choreonoid.