s-nakaoka/choreonoid #185: Video Tutorials/Documentation Search Function

Nbot07, 2017-12-16:
I noticed that many developers have a difficult time using the choreonoid documentation. In particular it seems that a search function or video tutorials would facilitate use. Are there video tutorials available for Choreonoid e.g. https://www.youtube.com/watch?v=3_kPZgoJl8I

s-nakaoka, 2017-12-20:
For the choreography function, there are tutorial videos:
http://choreonoid.org/ja/tutorial-videos.html
However, Japanese language is used in the videos.

Now we have been conducting a development project for a competition held in World Robot Summit, and we will improve manuals. First Japanese manuals will be improved, and then the manuals will be translated into English.

Nbot07, 2017-12-20:
Ok, may I share the video tutorials at http://choreonoid.org/ja/tutorial-videos.html on youtube? 
I believe this will allow it to reach a greater audience and reach them faster.

s-nakaoka, 2017-12-30:
AIST prohibits employees from uploading videos to Youtube. I think that the videos such as tutorials on Choreonoid should be able to be freely uploaded by a staff, so I will consult with the department in charge.

Nbot07, 2017-12-30:
Awesome, thanks!