s-nakaoka/choreonoid #188: モデルのいる場所の座標を表示、テキスト化したい。

HiroshiIzumitani, 2017-12-19:
こんにちは。

choreonoid上にあるモデルの位置の座標をテキストファイルとして読み取りたいのですが、そのような機能は実装されているのでしょうか？

おねがいします。

s-nakaoka, 2017-12-20:
モデルの位置と姿勢はボディ／リンクビューで確認することができますが、これではダメでしょうか？
http://choreonoid.org/ja/manuals/latest/handling-models/pose-editing.html#model-body-link-view

それとも、モデルの動き（軌道）の座標をファイルに出してほしいということでしょうか？
その場合、何をしたときの動きになりますか？シミュレーションしたときの動きでしょうか。

HiroshiIzumitani, 2017-12-21:
フィールドのどの位置（座標）に任意の物体があるかなどを記録する機能を実装したいと考えています。そこでシミュレーションを実行した時に、モデルの動き（軌道）の座標をファイルに出す機能について知りたいです。

Takuro-Sasaki, 2017-12-22:
1. シミュレーション実施
2. アイテムツリーでコントローラーやBodyItemの下にできているBodyMotionItemを選択
3. "ファイル" - "選択アイテムの保存"でファイルに出力

出力したファイルに時刻ごとのルートリンクの座標が載っています。
http://choreonoid.org/ja/manuals/latest/basics/item.html#id6

HiroshiIzumitani, 2017-12-22:
BodyMotionItemはもともとサンプルなどであるものでしょうか？アイテムツリーの中に見つかりません。

s-nakaoka, 2017-12-22:
こちらをご覧ください。
http://choreonoid.org/ja/manuals/latest/simulation/execution-and-playback.html#simulation-result-item-output

HiroshiIzumitani, 2017-12-25:
これで何のことかわかりました。ありがとうございます。

s-nakaoka, 2017-12-30:
ファイルフォーマットの解説を書きました。
http://choreonoid.org/ja/manuals/latest/reference/body-motion.html

closed 2018-05-14 by s-nakaoka