s-nakaoka/choreonoid #189: ビルドに失敗します

m-asahi, 2017-12-21:
Ubuntu14.04環境で最新のコミットをcloneしてビルドすると、src/Util/Vector3Seq.cppで以下のエラーが出てビルドが通りませんでした。（cmakeのオプションは何もつけていません）

履歴をたどってみたところ、12/19 1:57のコミット 12238ffb67d3a40196dc491c3bba8d43a0bce304 からのようです。

/home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp: In member function ‘virtual bool cnoid::Vector3Seq::doReadSeq(const cnoid::Mapping*, std::ostream&)’:
/home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp:63:10: error: no matching function for call to ‘cnoid::GeneralSeqReader::read(const cnoid::Mapping*&, cnoid::Vector3Seq* const, cnoid::Vector3Seq::doReadSeq(const cnoid::Mapping*, std::ostream&)::__lambda1)’
         });
          ^
/home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp:63:10: note: candidates are:
In file included from /home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp:10:0:
/home/asahi/workspace/choreonoid/src/Util/GeneralSeqReader.h:139:10: note: template<class SeqType, typename std::enable_if<(std::is_base_of<cnoid::AbstractSeq, SeqType>::value && (! std::is_base_of<cnoid::AbstractMultiSeq, SeqType>::value))>::type*& <anonymous> > bool cnoid::GeneralSeqReader::read(const cnoid::Mapping*, SeqType*, std::function<void(const cnoid::Listing&, int, typename SeqType::value_type&)>)
     bool read(
          ^
/home/asahi/workspace/choreonoid/src/Util/GeneralSeqReader.h:139:10: note:   template argument deduction/substitution failed:
/home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp:63:10: note:   ‘cnoid::Vector3Seq::doReadSeq(const cnoid::Mapping*, std::ostream&)::__lambda1’ is not derived from ‘std::function<void(const cnoid::Listing&, int, typename SeqType::value_type&)>’
         });
          ^
In file included from /home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp:10:0:
/home/asahi/workspace/choreonoid/src/Util/GeneralSeqReader.h:183:10: note: template<class SeqType, typename std::enable_if<std::is_base_of<cnoid::AbstractMultiSeq, SeqType>::value>::type*& <anonymous> > bool cnoid::GeneralSeqReader::read(const cnoid::Mapping*, SeqType*, std::function<void(const cnoid::ValueNode&, typename SeqType::value_type&)>)
     bool read(
          ^
/home/asahi/workspace/choreonoid/src/Util/GeneralSeqReader.h:183:10: note:   template argument deduction/substitution failed:
/home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp:63:10: note:   ‘cnoid::Vector3Seq::doReadSeq(const cnoid::Mapping*, std::ostream&)::__lambda1’ is not derived from ‘std::function<void(const cnoid::ValueNode&, typename SeqType::value_type&)>’
         });
          ^
/home/asahi/workspace/choreonoid/src/Util/Vector3Seq.cpp:64:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^

s-nakaoka, 2017-12-21:
ご報告いただきありがとうございます、確認が不十分だったようで申し訳ありません。
新たに定義したテンプレート関数について、コンパイラのバージョンによってはテンプレートパラメータを明示的に与えなくてはならなかったようで、そのように修正し、Ubuntu14.04でもコンパイルできることを確認しました。
c45d996c85a862412120c5fab76857383a3b61c9

m-asahi, 2017-12-22:
Ubuntu 14.04でビルド通りました。ありがとうございました。

closed 2017-12-22 by m-asahi