s-nakaoka/choreonoid #191: using alternate euler sequences to represent rotation

Nbot07, 2017-12-31:
Is there a way to change the sequence for the euler angles from ZYX to XYZ or other formats in choreonoid?

s-nakaoka, 2018-01-09:
No, currently there is no such a function. I think that the ZYX euler angles (roll-pitch-yaw expression) is usually used in the robotics community. If you really want the XYZ euler angles, we can implement it. 
Or you can use quaternion by checking the "option" - "BodyLink view" - "Use Quaternion" in the main menu.

Nbot07, 2018-01-09:
I really want XYZ euler angles. Please implement it if time allows. Also, would you mind commenting the code to help novices understand how to implement other euler sequences or representations of rotation?

s-nakaoka, 2018-01-09:
> I really want XYZ euler angles. Please implement it if time allows. 

OK. 
If possible, can you tell me the reason why you need XYZ euler angles?

> Also, would you mind commenting the code to help novices understand how to implement other euler sequences or representations of rotation?

I learned about the Euler angle in the following textbooks.
https://www.amazon.com/Introduction-Robotics-Mechanics-Control-Craig/dp/8131718360/ref=sr_1_3?ie=UTF8&qid=1515475575&sr=8-3&keywords=robotics+craig

Nbot07, 2018-01-09:
While developing a .body file for the InMoov humanoid robot, I discovered that positioning the stl files using the XYZ sequence is the most intuitive for new developers like myself. Admittedly, this may be due to our confusion regarding euluer sequences. Which if why a function that switches between representations might beneficial, since different developers or organizations may have a different sense of what is intuitive to them. Since we set the rotations of the parts using the XYZ sequence with a python script whose algorithm I augmented from http://www.euclideanspace.com/maths/geometry/rotations/conversions/eulerToAngle/index.htm we haven't been able to understand the Body/Link view's ZYX output. So, we would like to have the XYZ sequence to be able to tell the current rotation of a part.