s-nakaoka/choreonoid #193: モデル同士の衝突情報の取得

ikeyasu, 2018-01-06:
シミュレーションのタスクを達成したかを確認するため、モデル同士の衝突情報の取得をしたいとおもっています。

AISTSimulatorのregisterCollisionHandler を使おうとしたのですが、どうもうまくいきませんでした。（Materialの設定はしたのですが。。）

不躾な質問で恐縮ですが、何か簡単に取得する方法は無いでしょうか？例えばPythonから、2つのItem同士が衝突しているかを検知できたりすると良いのですが。。

s-nakaoka, 2018-01-09:
> 不躾な質問で恐縮ですが、何か簡単に取得する方法は無いでしょうか？例えばPythonから、2つのItem同士が衝突しているかを検知できたりすると良いのですが。。

もちろん出来ないことはないのですが、（Pythonラッパも含めて）今用意されているAPIだけでこれを行うのは難しそうです。一番良いのは、Body間もしくはLink間の衝突状況が変わる度にシグナルを発行するようなサブシミュレータ（SubSimulatorItemを継承して実装）を作成して、そのシグナルをPython側で拾うようにすることなのですが、実装方法を説明するのは少し大変ですし、周辺APIの整備も必要そうなので、サンプルを作成するのがよいかと思いました。今少し立て込んでおりまして、サンプルを作成するまでしばらくお待ちいただけますでしょうか。

ikeyasu, 2018-01-09:
@s-nakaoka ご返信、ありがとうございます。その後試行錯誤致しまして、以下のような形で実装をトライしています。が、Pythonからのコールバックでクラッシュしてしまい、これの原因が分からず、調査中です。。

ブランチ https://github.com/ikeyasu/choreonoid/tree/gym
diff https://github.com/s-nakaoka/choreonoid/compare/master...ikeyasu:gym

ext/tank 以下に、GymCollisionHandlerPlugin として実装しています。そこで、SubSimulatorItemを継承し、GymCollisionHandlerItem を実装し、ここで、Collisionの情報を得ています。

ソースの差分が多いですが、Tutorialのコードやその他の試行錯誤も含んでいます。
やっていることは、単に、AISTSimulator#registerCollisionHandler() のコールバックが呼ばれたら、PythonのCallbackを呼ぶようにしているだけです。
が、PythonのCallbackを呼ぶ時点で、Segmentation faultが起こっています。おそらく、Choreonoidの話ではなく、pybind11 の話だと思いますが。。

s-nakaoka, 2018-01-09:
ざっと見ただけなので見当違いかもしれませんが、まずコメントアウトしてありますが、Pythonコードの呼び出しにはcallLaterを使ったほうがよいです。また、Python関数の呼び出しでは、GIL(Global Interpreter Lock)というもののロックが必要です。Pythonはインタプリタ自体はマルチスレッド対応ではありませんので。

以上踏まえて、実際には以下のようにするとよいかと思います。

- コールバックを呼ぶためのシグナルをC++側で定義
- シグナルの呼び出しはcallLaterを使ってメインスレッドから呼ぶようにする
- シグナルのPythonラッパをpybind11側で定義。定義方法は他のシグナルのラッパを定義しているコードを参照ください。これを行うためのテンプレートがあります。
- そのようにして作成したシグナルのPythonラッパを使うと、GILのロックも自動的に行われます。
- あとはPython側で、シグナルのconnect関数を使って適当な関数に接続してください。シグナルをコールバック関数にconnectするサンプルはsample/python/ShakeBodies.pyを参照してください。

すみません、今時間がないのでとりあえずこのへんで。

s-nakaoka, 2018-01-09:
ちなみにシグナルの定義は、Signal<void()> みたいなシンプルな型でしたら、Utilモジュールでいくつか定義されているので、新たに定義する必要はありません。ただシグナルの引数に干渉に関する情報(link1, link2, collisionsとか）を含めるとすると、新たに定義する必要がありますね。あとその場合collisionsとかの型のラッパをまだ定義していなかったかも。あとcollisionsとかの(C++でのオリジナルの）型の定義について、あまり整理できていないので、今後変更するかもしれません。

ikeyasu, 2018-01-09:
@s-nakaoka 詳細なコメント、ありがとうございます。まだコードの整理ができていませんが、SignalでCallbackが実装できました。
https://github.com/s-nakaoka/choreonoid/compare/master...ikeyasu:gym

もう少しコードを整理したら、こちらにリンクを貼って、Closeにしたいと思います。

ikeyasu, 2018-01-15:
以下に整理したコードを書きました。
同様のニーズがある方は参考にして頂ければと思います。
https://github.com/s-nakaoka/choreonoid/compare/master...ikeyasu:collision-signal-plugin

closed 2018-01-15 by ikeyasu