s-nakaoka/choreonoid #197: AGXSimulatorで面が閉じていないstlファイルをモデルに使う場合の摩擦係数の設定

taka40606, 2018-01-12:
お世話になります．
環境ファイルとしてbodyファイルを作り，その中のelementsの中でstlファイルを読み込んでおり，
環境とロボットの間の摩擦係数をyamlファイルで設定してシミュレーションを行っております．
AISTSimulatorでは設定した摩擦係数が有効になっているのですが，
AGXSimulatorではどのような摩擦係数を設定しても摩擦係数が0のような動作をしてしまいます．
stlファイルのモデルの表面に穴が空いてしまっていることが原因であると考えていますが，
AGX Dynamicsでのシミュレーションで面が閉じていない場合でも摩擦係数を設定できるようにすることは可能でしょうか．

以下，プロジェクトファイルを開いたときに出る端末への表示です．
面が閉じていた場合は，この表示が出ず，摩擦係数も設定されます．

$ bin/choreonoid
Trimesh creation warnings for source 'empty':
13263 vertices are used 1 times only; should be at least 3 times.
13263 edges are used once instead of twice.
No single edge with half edge structure found. Collision detection will work badly.
Computed negative volume.
Is right triangle orientation/winding set? 
Use this trimesh only for static bodies; consider fixing it.

For details, look up the documentation about usage of agxCollide::Trimesh.
Trimesh creation warnings for source 'empty':
3252 vertices are used 1 times only; should be at least 3 times.
3252 edges are used once instead of twice.
No single edge with half edge structure found. Collision detection will work badly.

For details, look up the documentation about usage of agxCollide::Trimesh.
AGXDynamicsPlugin:INFO contact material [Tracks OilWetGround]
AGXDynamicsPlugin:INFO contact material [Ground Tracks]
AGXDynamicsPlugin:INFO contact material [default default]
AGXDynamicsPlugin:INFO contact material [Ground agxMat5]
AGXDynamicsPlugin:INFO contact material [Ground agxMat1]
AGXDynamicsPlugin:INFO contact material [Ground agxMat2]
AGXDynamicsPlugin:INFO contact material [Ground agxMat3]
AGXDynamicsPlugin:INFO contact material [Ground agxMat4]

susaikum, 2018-01-15:
面が閉じていない場合でも摩擦係数を設定できるようにすることは可能です。
摩擦係数が有効にならないのはstlファイルのモデルの表面に穴が空いてしまっていることが原因か
とご指摘されておりますが、こちらに関連はない可能性が高いです。
おそらく、摩擦係数の記述方法が誤っているのではないかと推察します。

AGXSimulatorで扱う物性はマテリアルファイルを利用しております。
デフォルトで読み込まれるファイルは`choreonoid/share/default/materials.yaml`です。
別途ご自身でファイルを用意し、Worldアイテムのプロパティから読み込ませることも可能です。
こちらに、Material(剛体の物性)とContactMaterial(剛体間の物性)を定義し、
ボディファイルに定義したMaterialを設定することで摩擦係数や反発係数を変更することができます。
サンプルとして、ディレクトリ`choreonoid/sample/AGXDynamics`の
- プロジェクトファイル: agxMaterialSample.cnoid
- ボディファイル: agxMaterialSample.body
- マテリアルファイル: agxSampleMaterials.yaml

をご確認されると良いと思います。

各パラメータの詳細については
[Choreonoid 開発版 ドキュメント->AGX Dynamicsプラグイン->物理マテリアル](http://choreonoid.org/ja/manuals/latest/agxdynamics/agx-material.html)
をご確認ください。

改善しないようでしたら、疑似コードでも良いのでどのような記述、設定をされているかご提示いただければと思います。

taka40606, 2018-01-16:
ご回答有り難うございます．

サンプルのagxMaterialSample.cnoidのfloorをstlファイルに変えてみたところ，
stlファイルの環境とboxの間の摩擦係数を設定することができました．

また，いくつか設定を変えて試したところ，摩擦が設定できる条件がありましたので報告いたします．
私の試したプロジェクトは，
・環境のbodyファイル（stlファイルとprimitiveのBox）
・ロボットのbodyファイル（primitiveを使ったロボットモデル）
・物性を設定するyamlファイル（AGX Dynamicsのサンプル内にあるファイル）
を読み込んで，AGXSimuratorでシミュレーションを行います．

１．yamlファイルのfrictionModeのソルバをdirectに変えると常に摩擦係数の設定が可能．

２．yamlファイルのfrictionModeのソルバをsplitにすると，
環境のbodyファイル内でBoxより先にstlファイルのモデルを読み込んだ場合，Box上でもstlファイルのモデル上でも摩擦が効いているが，stlファイルのモデルとBoxの両方に接地している場合，摩擦係数が0のような動作をする．（bodyファイルの内容は下記のとおりです）
※環境のbodyファイル内でstlファイルのモデルより先にBoxを読み込んだ場合，Box上では摩擦が効いているが，stlファイルのモデル上では摩擦係数が0のような動作をする．

agxMaterialSample.body（環境のbodyファイル stlファイルのモデルを読み込んだ場合）
```
format: ChoreonoidBody
formatVersion: 1.0
angleUnit: degree
name: agxMaterialSample

links:
  -
    name: floor
    translation: [ 0, 0, -0.1 ]
    rotation: [1, 0, 0, -0 ]
    jointType: fixed
    material: Ground
    elements:
      -
        type: Shape
        translation: [ 0, 0, 0 ]
        geometry: { type: Resource, uri: "mesh.stl" }
        appearance: { material: { diffuseColor: [ 1 , 0, 0 ] }}
  -
    name: floor2
    parent: floor
    translation: [ 0, 0, -0.1 ]
    rotation: [1, 0, 0, -0 ]
    jointType: fixed
    material: Ground
    elements:
      -
        type: Shape
        translation: [ 0, 0, 0 ]
        geometry: { type: Box, size: [ 10.0, 10.0, 0.2 ] }
        appearance: { material: { diffuseColor: [ 0, 0, 1 ] }}
```

agxMaterialSample.body（環境のbodyファイル Boxを読み込んだ場合）
```
format: ChoreonoidBody
formatVersion: 1.0
angleUnit: degree
name: agxMaterialSample

links:
  -
    name: floor
    translation: [ 0, 0, -0.1 ]
    rotation: [1, 0, 0, -0 ]
    jointType: fixed
    material: Ground
    elements:
      -
        type: Shape
        translation: [ 0, 0, 0 ]
        geometry: { type: Box, size: [ 10.0, 10.0, 0.2 ] }
        appearance: { material: { diffuseColor: [ 1 , 0, 0 ] }}
  -
    name: floor2
    parent: floor
    translation: [ 0, 0, -0.1 ]
    rotation: [1, 0, 0, -0 ]
    jointType: fixed
    material: Ground
    elements:
      -
        type: Shape
        translation: [ 0, 0, 0 ]
        geometry: { type: Resource, uri: "mesh.stl" }
        appearance: { material: { diffuseColor: [ 0, 0, 1 ] }}
```

susaikum, 2018-01-16:
ご確認ありがとうございます。
agxMaterialSample.bodyに追加でstlファイルを読み込ませ、摩擦係数が0のような動作をするか確認いたしましたが、再現ができておりません。
テスト内容を添付いたしますので、ご確認いただけますでしょうか。
[agxMaterialSTLTest.zip](https://github.com/s-nakaoka/choreonoid/files/1634571/agxMaterialSTLTest.zip)

taka40606, 2018-01-17:
テストで確認してみたところ、私の環境でも正常に動作しました。

私の方でも摩擦が0になるプロジェクトを作成しましたので確認して頂いてもよろしいでしょうか。
[agxMaterialSTLTest2.zip](https://github.com/s-nakaoka/choreonoid/files/1637412/agxMaterialSTLTest2.zip)

susaikum, 2018-01-22:
確認いたしました。
摩擦が0のような動作をしているように見受けられませんが、どの部分でしょうか。