s-nakaoka/choreonoid #198: Windows環境でのAGXDynamics Pluginの使用について

ytazz, 2018-01-19:
Windows環境でAGXdynamicsプラグインをセットアップする方法に関する情報が
見当たりませんでしたのでここに投稿致します。

以下の手順によりAGXDynamicsプラグインを利用しようとしているのですが、
起動時に添付画像のようなプラグインロードエラーが出力されてしまいます。
サンプルでシミュレーションを実行しても応答がありません。

[手順]
・インストーラ（agx-setup-2.21.1.1-x64-VS2015-double.exe）を実行
・インストールディレクトリ（C:\Program Files\Algoryx）にライセンスファイル（agx.lib）を置く
・ライセンス認証用ハードウェアキーを接続
＊この段階でAGX添付のチュートリアルサンプルの動作は確認できました

・Choreonoidのソースをgithubから入手
・CMakeでAGX関係のオプションを有効化
・VS2015でビルドしてインストール（C:\Program Files\Choreonoid\bin）

・AGXインストールディレクトリのsetup_env.batの以下を編集
　＊自分の環境にPython3.6がインストール済のため
　set AGX_USE_EXTERNAL_PYTHON=1
　set PYTHONHOME=c:\Program Files\Python36

・コマンドプロンプト起動
・C:\Program Files\Algoryx\AGX-2.21.1.1\setup_env.bat
　を実行
・C:\Program Files\Choreonoid\bin\choreonoid.exe
 を実行
 
＊DLL検出自体はされているのにロードエラーとなっているので
　依存DLLが見つけられていないのかと思うのですが、
　AGXのbinディレクトリへのパスは通っているので他に原因が思いつきません。

よろしくお願い致します。
[image]

hattorishizuko, 2018-01-25:
ご指摘ありがとうございます。対応が遅くなりまして申し訳ありません。
画面を拝見しますと、エラー表示の後に、AGX***プラグインが読み込まれました。
と表示されているので、dllのロードはできていると思われます。
このエラー表示は、プラグイン同士の依存関係で表示されるもので、AGX****プラグインが例えばBodyプラグインに依存してる場合、最初にロードしようとした時にはBodyプラグインが見つからずエラーとなり、Bodyプラグインをロードした後、再度AGX****プラグインを読み込んで、成功しているということです。
従いまして、サンプルが動作しない原因は他にある思われます。
http://choreonoid.org/ja/manuals/latest/agxdynamics/install/install.html
のページにも記載しておりますが、現在Ubuntu16.04 LTS x64にて動作を確認している状態でWindowsでの動作確認はできていない状態です。
今後、Windowsにも対応していく予定ですので、今しばらくお待ちいただけますようよろしくお願いします。

s-nakaoka, 2018-01-25:
すみません、AGXDynamicsPluginについて、まだ開発中のところもあり、いろいろとご不便をおかけしております。

使用するAGXのバージョンについて注意がありまして、現在正常に動作するのはバージョン2.20.0.1となっております。マニュアルに書かれているバージョンや、最新のバージョンでは、うまく動作しないようですので、ご注意ください。マニュアルについては今後更新いたします。

ダウンロードは以下から可能です。
https://www.algoryx.se/download/?id=1348

s-nakaoka, 2018-01-25:
この件マニュアルに追記されました。
http://choreonoid.org/ja/manuals/latest/agxdynamics/install/agx-platform.html

ちなみに先ほどgithubにアップしたChoreonoidの最新版では、最新のAGXバージョンにも対応しているとのことです。

ytazz, 2018-01-25:
Choreonoid最新版にてWindows+VS2015 x64環境で
TankJoystickAGX_Floorサンプルの動作を確認できました。

ご対応ありがとうございます！

closed 2018-02-03 by ytazz