s-nakaoka/choreonoid #199: Compilation error when attempting to compilie CnoidCorbaPlugin from source

CheekyMonkey, 2018-02-19:
Hi

I'm seeing the following compilation error when attempting to compile the CnoidCorbaPlugin from the github source:

> 1>------ Build started: Project: CnoidCorbaPlugin, Configuration: Release x64 ------
> 1>  Building Custom Rule C:/projects/choreonoid/src/CorbaPlugin/CMakeLists.txt
> 1>  CMake does not need to re-run because C:/projects/choreonoid/build/src/CorbaPlugin/CMakeFiles/generate.stamp is up-to-date.
> 1>  Generating ../../../share/locale/ja/LC_MESSAGES/CnoidCorbaPlugin-1.7.mo
> 1>  Copying MessageView.idl to C:/projects/choreonoid/include/cnoid/corba
> 1>          1 file(s) copied.
> 1>  Generating the C++ stubs and skeletons of MessageView.idl
> 1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(171,5): error MSB6006: "cmd.exe" exited with code -1073741515.
> ========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

Environment:
Windows 10 
Microsoft Visual Studio Community 15 Update 3
Choreonoid (latest from github: https://github.com/s-nakaoka/choreonoid/ as at commit  41133c6)
Boost (boost_1_60_0-msvc-14.0-64)
QT (qt-unified-windows-x86-3.0.2-online)
Eigen  (version 3.3.4)
Cmake (version 3.9.5)

Steps to reproduce (http://choreonoid.org/en/manuals/latest/install/build-windows.html)
cmake > (configure and generate: success) > enable: Boost, Eigen, QT
Visual Studio > build solution

Verbose compiler output attached for reference
[choreonoid compilation error.txt](https://github.com/s-nakaoka/choreonoid/files/1736226/choreonoid.compilation.error.txt)

Any guidance to overcome this error?

I did read in your compilation [guidance](http://choreonoid.org/en/manuals/latest/install/build-windows.html) that there were challenges with compiling using Microsoft Visual Studio Community 15 Update 3 and but that there was success with Microsoft Visual Studio Community 15 Update 2.

Any tips on how to roll  back "MVSC update 3", if this is the way to go?

ytazz, 2018-03-02:
Hi I am another user, and strangely I did not encounter the error you reported,
 using VS community 2015 Update 3.
Newest source from github.

CheekyMonkey, 2018-03-02:
Hi @ytazz

Are you able to share your cmake cache file for comparison?

ytazz, 2018-03-02:
hi CheekyMonkey

there you go

[CMakeCache.txt](https://github.com/s-nakaoka/choreonoid/files/1773939/CMakeCache.txt)

CheekyMonkey, 2018-03-08:
Thanks @ytazz

Having moved to another laptop, I’m now able to complete the compile without issue (exact same software versions) 

Go figure...

Please close this issue

closed 2018-05-14 by s-nakaoka