s-nakaoka/choreonoid #201: Generating Sequence Files

Nbot07, 2018-02-28:
How are .seq files generated in Choreonoid?

mohamadmahdavian, 2019-08-02:
Did you find the answer to this question dear @Nbot07 ?

Nbot07, 2019-08-04:
No, I haven't found the answer yet.

Nbot07, 2019-09-25:
In the optional plugins overview subsection of the installation section of the latest manual, there is a build configuration option named BUILD_POSE_SEQ_PLUGIN that 
"builds PoseSeq plug-in, which provides provides the choreographic function by key pause. It is configured to ON by default." https://choreonoid.org/en/manuals/latest/install/options.html

Let me know if that helps, and the other things you have tried.

Nbot07, 2019-10-02:
@mohamadmahdavian 

There appears to a method for creating sequence files outlined in the "edit sample motion" subsection of the "choreography section" of the tutorial part of the main Japanese website for choreonoid. It may be useful to use a browser, like chrome, with autotranslation enabled to view the page. 
https://choreonoid.org/ja/choreograph-tutorial/editSampleMotion.html

Let me know if that helps!

Nbot07, 2019-10-18:
@mohamadmahdavian 

This may also be useful https://choreonoid.org/en/manuals/latest/reference/body-motion.html