s-nakaoka/choreonoid #212: TimeBar stops when there is no dynamic object (only static object)

yosuke, 2018-09-27:
シミュレーション開始ボタンを押すと通常であればタイムバーがシミュレーション時刻に従って更新されますが、ワールドにstaticなオブジェクト（floorなど）しかない場合、少しだけ進んだあとで止まってしまいます。

シミュレーション停止ボタンを押すと時間経過は正しくログに表示されるので、シミュレータではなく、タイムバー（とおそらくロガー）の問題と思われます。

s-nakaoka, 2018-09-27:
staticなオブジェクトしかない場合、物体の動作ログ(BodyMotionItemに格納されるもの）が出力されません。タイムバーやアニメーションの機能は基本的に動作ログを対象に機能しますので、動作ログがないとアニメーションも出来ないということになります。

ただ、実際にはシミュレーションは内部で動作しているのだとすると、そことの食い違いがあるため、分かりにくいですね。staticなオブジェクトしか無い場合は、シミュレーションも開始できないようにするのがよいかもしれませんね。staticなオブジェクトだけでセンサも何もない場合はそもそもシミュレーションする必要はありませんので。

ただしモデル自体はstaticでも、それが何らかのデバイス（センサ等）をもっている場合は、そちらのシミュレーションを行う必要がありますし、デバイスの状態もログに保存されるので、タイムバーも機能するべきですね。今回はそのようなことをしようとしているが、実際にはうまく動作しないということでしょうか？そうでしたらそこについては修正したいと思います。

yosuke, 2018-09-27:
ROSの流儀だと、シミュレーターはstaticなオブジェクトのみをロードした状態で立ち上がって＆立ち上がった状態から時間は進んでいて、後からAPIを使ってロボットなどdynamicなオブジェクトをロードさせるので、その場合だとあれれ、となります。

立ち上げ時からロボットをロードしてしまえば良い話なので、私についてはそんなに対応を急いでいません。

yosuke, 2018-09-27:
あと、pythonスクリプトを使ってstaticなオブジェクトをアニメーションぽく動かしてしまうパターンもあるかもしれないですね。その場合だとstaticなオブジェクトであっても動作ログに姿勢を保存してほしい（この点についてもただの思いつきで対応は全く急いでいません）。