s-nakaoka/choreonoid #227: AGXSimulatorで Customizerが動作しない

YoheiKakiuchi, 2020-01-08:
シミュレーターをAISTSimulatorから、AGXSimulatorへ変更したとき（その他のモデルは変更しない）
AISTSimulatorでは動作していた CustomizerがAGXSimulatorでは動作していないように見えます。
AGXSimulatorでCustomizerを使うには何か設定等必要でしょうか？

s-nakaoka, 2020-01-17:
Customizerというのは関節軸に与えるvirtualJointForcesのことだと思うのですが、すみません、今確認したところAGXSimulatorではまだこの要素が取り込めていませんでした。

カスタマイザで記述したvirtualJointForcesは、BodyクラスのsetVirtualJointForcesによって設定されます。

Choreonoidのsrc以下で

grep -r setVirtualJointForces

とすると、この関数の使用箇所が分かります。

Body/DyWorld.cppやODEPlugin/ODESimulatorItem.cpp等で使用している箇所が、これをシミュレーション中にセットしているところです。

AGXDynamicsPluginでも適切な箇所に同様のコードを挿入することにより、同じことが実現できるかと思います。

すみません、今のところはコードの修正をする時間がとれないので、上記の解決方法の提示のみとなりますが、ご了承ください。

s-nakaoka, 2020-01-17:
ちなみにAGXSimulatorについては、そもそもAGXが同様の機能（関節にコンプライアンスを設定する）をもっているので、そのパラメータを記述した方が楽で確実かと思います。
使用可能なパラメータはマニュアルの以下のページで説明しています。
https://choreonoid.org/ja/manuals/latest/agxdynamics/agx-body.html
complianceとかdamping関連のパラメータを調整することで、カスタマイザと同様の挙動を実現できるかもしれません。

s-nakaoka, 2020-06-24:
@YoheiKakiuchi

こちらAGXのパラメータ設定では対応できませんでしたでしょうか？Customizerの利用目的が仮想関節でダンパを実現することでしたら、AGXの関節パラメータである程度対応できるのではないかと思うのですが・・・。

YoheiKakiuchi, 2020-06-24:
@s-nakaoka 
本件確認できておらず、申し訳ありません。
AGXの関節パラメータについて検討してみたいと思います。

closed 2020-06-24 by YoheiKakiuchi