s-nakaoka/choreonoid #230: Use of removed function install_external_libraries in .cmake files

AlgoryxJosef, 2020-02-27:
I noticed that the cmake function "install_external_libraries" was removed in commit 9b86928, but is still used in some of the CMakeLists.txt files, giving cmake configuration errors.

For example ../src/BulletPlugin/CMakeLists.txt (and a few others) tries to call install_external_libraries (line 82).

s-nakaoka, 2020-02-28:
Thank you for pointing out this problem.
When I modified the main CMakeLists.txt, I couldn't fix these plugins. I'm sorry about that.
In the new main CMakeLists.txt, install_external_libraries is replaced with install_runtime_dlls, and I have replaced the old function with the new one for the relevant plugins.
4de96c888559e56f022e4961a1b1e80049d6fd82
Probably the plugins can be compiled now.

AlgoryxJosef, 2020-03-02:
Great!

closed 2020-03-02 by AlgoryxJosef