s-nakaoka/choreonoid #231: Using the bullet plugin crashes the application

AlgoryxJosef, 2020-03-02:
Starting any simulation with the Bullet plugin crashes the application without any error messages.

To reproduce: build latest Choreonoid (master) with the Bullet-plugin. Bullet version is 2.83.7.
Open any project, e.g. Falling Boxes, select the Bullet simulator and press play.
The application crashes without any error messages.

This has been reproduced on both Windows and Linux.

s-nakaoka, 2020-06-24:
Thank you for letting us know about this issue. I'm sorry about the bug, but we don't use the Bullet very often now, and it may be difficult to fix it for the time being.  For reference, can you tell us what purpose the Bullet will be used for with Choreonoid?

In fact AGXDynamics is an excellent physics engine, and I think it's the best engine in the world. However, it is also a costly thing to use, Choreonoid needs to support free physics engines as well. There are several famous free physics engines including ODE, Bullet, PhysX, etc. , but I'm not sure which is the best engine Choreonoid should support. If you know something about that, please let me know it.

s-nakaoka, 2020-06-24:
Once we know which free physics engine is best for us, we would like to prioritize the development of the plugin for it.