s-nakaoka/choreonoid #239: No SensorVisualizer plugin found in Choreonoid simulator 1.7.0

kx-guan, 2020-04-27:
Hi, Dear everyone:
Thank you very much for discussion.
I got the following error messages when loading a project with the Choreonoid simulator 1.7.0:

============================
 SensorVisualizer "SensorVisualizer" を読み込み中
loading 
 cannot be found.
Retrying to load the file as a standard VRML file
 cannot be found.
loading failed.
 cannot be found.
============================

  I investigated the source of 1.7.0. I have not found a SensorVisualizer plugin. And there is not a ON/OFF flag for cmake configuration.
Could you give me any advice?
Thank you very much.
Best regards.

ssr-yuki, 2020-04-27:
Hi, @kx-guan 

`SensorVisualizer` is a basic function of Choreonoid.
You should check your project: `SensorVisualizer` must be a child item of the body item. Please see [here](https://www.choreonoid.org/en/manuals/latest/simulation/sensor-visualization.html#sensor-visualizer-item).

Regards.

kx-guan, 2020-04-28:
Hi, @ssr-yuki 
  OK. Thank you very much for quick feedback.
Best regards.

closed 2020-06-24 by s-nakaoka