s-nakaoka/choreonoid #240: Can't able to use Virtual Joystick

DhimanAirao, 2020-04-27:
In the version 1.7.0 I tried to use Tank model and make it walk through Virtual Joystick But there is no response from the Robot Model then I tried different model but it didn't work in any model. Please help me to fix this.

ssr-yuki, 2020-04-27:
Do you surely use `TurretController2` as a controller of Tank?

If "Yes," could you explain the problem you are facing more in detail?

DhimanAirao, 2020-04-27:
Yes I am using TurretController2 as a controller of Tank. I also tried to make entire project myself even though it didn't work. Can you help me to fix this?

ssr-yuki, 2020-04-27:
Oh, something wrong.

I guess you tried [this tutorial](https://www.choreonoid.org/en/manuals/latest/simulation/tank-tutorial/step3.html#controller-source-code), right?

I want to see console messages. Are there any warnings or errors?

DhimanAirao, 2020-04-27:
[image]
This is the screenshot and there is also other problem that turret rotates continously.Please help me?

ssr-yuki, 2020-04-27:
Let me summarize your problems:
-  Your virtual joystick does not work with `TurretController2`.
- The turret rotates continuously with `TankTutorial_TurretController2`, and WITHOUT any joystick command.

Is that all?

DhimanAirao, 2020-04-28:
Yes, sir. Please help me to fix it.

DhimanAirao, 2020-04-28:
Also I can't able to load fire.body file located in sample/wrs2018/model it shows error as show below

Error is:
The node type "FireDevice" is not defined. at line 20, column 9. -> failed.

How to fix this?

DhimanAirao, 2020-04-29:
@s-nakaoka sir help me to figure out the 3 problems listed above.

s-nakaoka, 2020-05-01:
You can make FireDevice available by enabling BUILD_SCENE_EFFECTS_PLUGIN in the CMake configuration.

For the above two problems, I'm not sure why the problems occur.  Did you correctly follow the following manual pages?
https://choreonoid.org/en/manuals/latest/simulation/tank-tutorial/index.html

It may be better to use the latest development version of Choreonoid.
https://choreonoid.org/en/manuals/latest/install/build-ubuntu.html#development-version

DhimanAirao, 2020-05-01:
Sir, I tried 3 different version of chorenoid but it didn't work in ny version. I follwed the tutorial perfectly . Initially I tried to use the inbuilt files but It didn't work. So secondly I tried ti create my own project but still it didn't work.

s-nakaoka, 2020-05-01:
I remembered that I had such problems while using a virtual machine. If you use a virtual machine, please try to execute Choreonoid on the native Ubuntu environment.

DhimanAirao, 2020-05-01:
No sir I have dual boot in my Laptop I ain't using Virtual Machine

s-nakaoka, 2020-05-01:
I think it's difficult to answer to your questions because you don't provide enough information to solve the problems. I'm afraid it may be better to refer to the following page:
http://www.catb.org/~esr/faqs/smart-questions.html

DhimanAirao, 2020-05-01:
Sir, I had provided every information that I m able to provide. I uploaded the screenshot of what's happening while using the virtual joystick in chorenoid.

s-nakaoka, 2020-05-01:
I also recommend you to test your joystick by using the jstest command. You can install it by apt install joystick.

s-nakaoka, 2020-05-01:
Sorry, you used a virtual joystick. If some joystick is connected, the input from it is used

DhimanAirao, 2020-05-01:
Sir also when I open Tank.cnoid file present in Sample/Simplecontroller the Turret rotates continously without any input. How to fix this problem?

s-nakaoka, 2020-05-01:
I'm not sure why the problem occurs in your environment. I'm sorry about that.

DhimanAirao, 2020-05-01:
Thanks Sir for your reply. I will try to solve this problem.

DhimanAirao, 2020-05-01:
[image]

Sir, made 3 plugins on during make  BUILD_SCENE_EFFECTS_PLUGIN , BUILD_SCENEEDIT_PLUGIN and BUILD_SCENARIO_PLUGIN  but as you can see now fire.body is getting perfectly loaded but can't able to see it in the world.

DhimanAirao, 2020-05-01:
how to fix this?

s-nakaoka, 2020-05-07:
FireDevice itself does not show anything just by loading it into the scene. So there is nothing wrong with what is happening in your environment.

closed 2020-05-07 by s-nakaoka